adafruit/Adafruit_CircuitPython_ServoKit #7: AttributeError: module 'board' has no attribute 'I2C'

sumghai, 2019-03-22:
### Summary of Issue
Attempting to initialize the class object from `ServoKit` results in the following error:
```
Traceback (most recent call last):
  File "servo_test.py", line 3, in <module>
    kit = ServoKit(channels=16)
  File "/usr/local/lib/python3.5/dist-packages/adafruit_servokit.py", line 85, in __init__
    i2c = board.I2C()
AttributeError: module 'board' has no attribute 'I2C'
```
### Hardware
- Raspberry Pi 3 Model B
- Adafruit 16-Channel 12-bit PWM/Servo Driver - I2C interface - PCA9685
- (2x) GWS S03T STD Servos

### Software and Environment
- Raspbian Stretch (Debian 9, Release 2018-11-13)
- Python 3
- Adafruit GPIO 1.0.3
- Adafruit Blinka 1.2.6 (_I2C verified working with [blinkatest.py example file](https://learn.adafruit.com/circuitpython-on-raspberrypi-linux/installing-circuitpython-on-raspberry-pi)_)
- Adafruit CircuitPython ServoKit 1.1.0
- Installed and configured as per [the 16 channel servo driver tutorial](https://learn.adafruit.com/adafruit-16-channel-servo-driver-with-raspberry-pi/using-the-adafruit-library)

### servo_test.py
```
import time
from adafruit_servokit import ServoKit
kit = ServoKit(channels=16)

# Tuned for GWS S03T STD Servos
kit.servo[0].set_pulse_width_range(775, 2450)
kit.servo[1].set_pulse_width_range(775, 2450)

# Return servos to starting positions
kit.servo[0].angle = 0
kit.servo[1].angle = 180
time.sleep(1)

while True:
    for tgtAngle in range(0, 180, 1):
        kit.servo[0].angle = tgtAngle
        kit.servo[1].angle = 180 - tgtAngle
        time.sleep(0.1)
    for tgtAngle in range(180, 0, -1):
        kit.servo[0].angle = tgtAngle
        kit.servo[1].angle = 180 - tgtAngle
        time.sleep(0.1)
```

### Notes
- I previously had CircuitPython ServoKit working with the Python code sample and two servos one week ago
- This morning, I had to completely reformat and re-setup my Raspberry Pi due to an unrelated issue
- After reinstalling CircuitPython ServoKit and all dependencies from scratch, I found that my existing code no longer works and instead returned the titular error message, so I presume there was a change in adafruit_servokit.py between my two installations
- Examining adafruit_servokit.py, I noticed that line 85 was previously (828a46c66e685a27980cdfac6f11aa131739defe) defined as `i2c = busio.I2C(board.SCL, board.SDA)` instead of the current `i2c = board.I2C()`; perhaps this line may be of interest
- The blinkatest.py example currently still works because it uses the older `i2c = busio.I2C(board.SCL, board.SDA)`

abhiTronix, 2019-03-22:
@sumghai Bug confirmed. Tested on my Raspberry Pi 3 Model B.

kattni, 2019-03-22:
@sumghai @abhiTronix We were fixing a different I2C bug and introduced this one. This is fixed by https://github.com/adafruit/Adafruit_Blinka/pull/95. Please update Adafruit_Blinka to resolve the issue. I've done a new release - give it a few minutes to deploy to PyPI before trying to update.

To answer your question, @abhiTronix, to pass in different pins, you would instantiate the library as follows:
```
import busio
from adafruit_servokit import ServoKit
kit = ServoKit(channels=16, i2c=(busio.I2C(board.SCL, board.SDA)))
```
I have only tested that on a microcontroller, not on RasPi, but that is the idea of how to pass in different I2C pins.

closed 2019-03-22 by kattni

sumghai, 2019-03-22:
I can confirm that updating Adafruit Blinka to 1.2.7 resolves this issue on the RasPi.

Thanks @kattni!

hamwong, 2019-12-20:
HI, 

I am try to use below code and have same error
`    import board
    import busio
    import adafruit_tsl2561
    i2c = busio.I2C(board.SCL, board.SDA)
    sensor = adafruit_tsl2561.TSL2561(i2c)`

I am using Rpi 3B+ with  Adafruit-Blinka 3.3.4
`Version: 3.3.4
Summary: CircuitPython APIs for non-CircuitPython versions of Python such as CPython on Linux and MicroPython.
Home-page: https://github.com/adafruit/Adafruit_Blinka
Author: Adafruit Industries
Author-email: circuitpython@adafruit.com
License: MIT
Location: /usr/local/lib/python3.5/dist-packages
Requires: Adafruit-PlatformDetect, rpi-ws281x, spidev, RPi.GPIO, sysv-ipc, pyftdi, Adafruit-PureIO
`

I got below error
`Traceback (most recent call last):
  File "./TSL2561c.py", line 12, in <module>
    i2c = busio.I2C(board.SCL, board.SDA)
AttributeError: module 'board' has no attribute 'SCL'
`

caternuson, 2019-12-20:
You aren't importing `board`? But somehow it must be or the error would be different. What does this show?
```python
import board
dir(board)
```

hamwong, 2019-12-22:
> 
> 
> You aren't importing `board`? But somehow it must be or the error would be different. What does this show?
> 
> ```python
> import board
> dir(board)
> ```

Sorry, I didn't double check the code I post, I have , it's at first line and I miss to copy it and paste here.
but I am 100% sure import board is there, I add your line and it didn't show alarm but I found further when print out, it looks my board have some problem

`import smbus
import time
import board
dir(board)
import busio
import adafruit_tsl2561
print("board contents: ", dir(board))
i2c = busio.I2C(board.SCL, board.SDA)
sensor = adafruit_tsl2561.TSL2561(i2c)

python3 TSL2561c.py
board contents:  ['BaseDimension', 'Board', 'Dimension', 'Empty', 'Image', 'InfiniteDimension', 'Infinity', '_Empty', '_InfiniteDimension', '_Infinity', '__builtins__', '__cached__', '__doc__', '__file__', '__loader__', '__name__', '__package__', '__spec__', '_centred_coord', 'cornerposts', 'functools', 'imagefile_sprite', 'io', 'itertools', 'long', 'os', 'sys', 'text_sprite']
Traceback (most recent call last):
  File "TSL2561c.py", line 15, in <module>
    i2c = busio.I2C(board.SCL, board.SDA)
AttributeError: module 'board' has no attribute 'SCL'
`

ladyada, 2019-12-22:
looks like you have some other library that has created a board.py in your python collection, & you'll have to remove it

caternuson, 2019-12-22:
Yep. You somehow have this installed:
https://pypi.org/project/board/
instead of the `board` that comes with Blinka.

You can not install the `board` module individually. You need to do a complete Blinka install.

hamwong, 2019-12-22:
Thanks @caternuson  and @ladyada 
after remove board and reinstall Blinka, work as expect.

dayojohn16profile, 2024-11-27:
not solved

DvarfInkviz, 2025-06-03:
Solved my problem with [Adafruit_Python_Extended_Bus](https://github.com/adafruit/Adafruit_Python_Extended_Bus)