adafruit/Adafruit_CircuitPython_ServoKit #8: Facing angular movement issue.

cis-apoorv, 2019-05-02:
I'm trying to rotate the servo in a while loop by increment of 10 degrees on every iteration. I've set the acutation angle to 180 degrees, but when I tried to rotate the motor to 180 degrees it move upto a certain angle then stopped. I perform check on the angle then I found that whenever I set the angle of motor to 0 degrees it set the value of the servo to -0.36600244001626675.

**When I set servo angle to 0-degree**
```
>>> kit.servo[1].angle=0
>>> kit.servo[1].angle
-0.36600244001626675

```

**When I set servo angle to 180-degrees**
```
>>> kit.servo[1].angle=180
>>> kit.servo[1].angle
180.0

```
Can you please help me out with this?

dastels, 2019-05-02:
Servos can be somewhat inconsistent and your code often needs some tweaking. You might have to adjust the min and max pulse widths.  These guides might be useful: https://learn.adafruit.com/circuitpython-servo-tester, https://learn.adafruit.com/adafruit-motor-selection-guide/rc-servos, https://learn.adafruit.com/adafruit-arduino-lesson-14-servo-motors.

cis-apoorv, 2019-05-03:
Thanks a lot @dastels sir! It is working like a charm. I've set the minimum and maximum pulse to 500, 2500 and now I'm getting perfect angular movement in the servo.

closed 2019-05-03 by cis-apoorv