adafruit/Adafruit_CircuitPython_ServoKit #10: Use unused PCA9685 FeatherWing's pin as inputs.

bpjobin, 2019-10-10:
Hi, 
I'm driving two servo motors using a Huzzah Feather esp8266 and a FeatherWing 8-channels PCA9685 board. Since this is leaving me with unused pins on wing board, I'd like to know if it's possible to use them as input pins to catch data from NC/NO switches or ultrasonic distance sensors.

`
self._button = Pin(5, mode=Pin.IN, pull=Pin.PULL_UP)
`

If so, how would I define one of those pins?

Thank you

ladyada, 2019-10-10:
hiya, you can use anything other than SCL/SDA - note we don't support esp8266 on circuitpython anymore! if youre using micropython, check to see if they have any drivers for the pca9685 :)

closed 2019-10-10 by ladyada

bpjobin, 2019-10-10:
I'm new to I2C protocol. What do you mean by: 
`you can use anything other than SCL/SDA`
I'm using the latest version 3 of circuitpython released in august so it works for now. I'll make sure to ping the folks of micropython too.

ladyada, 2019-10-13:
hiya you can use any gpio for inputs, OTHER than SDA/SCL pins