adafruit/Adafruit_CircuitPython_ServoKit #12: A function to access the current servo angle

Chaitanya-Mehta, 2019-10-21:
If we can add a function to access the current angle of the servo that is commanded, the library will be much more useful. I am using PCA9685 in a robot and I need to access the current angle of the servo to command the next angle via increment.

ladyada, 2019-10-21:
that isn't supported but if you'd like to add it and submit a PR it would be welcome!

Chaitanya-Mehta, 2019-10-22:
Alright! I am not so experienced with programming, but I will surely try. Would you please let me know the class where I need to make necessary changes, and provide a link to that?

caternuson, 2019-10-23:
Maybe it's already in place?
<https://github.com/adafruit/Adafruit_CircuitPython_Motor/blob/master/adafruit_motor/servo.py#L109>
Would this work?
To set:
```python
kit.servo[0].angle = 180
```
To query:
```python
kit.servo[0].angle
```

ladyada, 2019-10-23:
@Chaitanya-Mehta please try that

Chaitanya-Mehta, 2019-10-28:
Yes. It works as the angle is defined as property. I found that way and also checked via implementation. So, it's working.

To get the last servo angle in the variable x;
x= kit.servo[0].angle

closed 2019-10-28 by Chaitanya-Mehta