adafruit/Adafruit_CircuitPython_ServoKit #13: Zero Division Error

OgiBalboa, 2019-10-31:
Hey There, i am trying to control a servo via PCA9685 and having this error from servo example scripts :
`Traceback (most recent call last):
  File "/home/pi/Desktop/tri.py", line 33, in <module>
    servo7 = servo.Servo(pca.channels[7])
  File "/home/pi/.local/lib/python3.5/site-packages/adafruit_motor/servo.py", line 103, in __init__
    super().__init__(pwm_out, min_pulse=min_pulse, max_pulse=max_pulse)
  File "/home/pi/.local/lib/python3.5/site-packages/adafruit_motor/servo.py", line 45, in __init__
    self.set_pulse_width_range(min_pulse, max_pulse)
  File "/home/pi/.local/lib/python3.5/site-packages/adafruit_motor/servo.py", line 49, in set_pulse_width_range
    self._min_duty = int((min_pulse * self._pwm_out.frequency) / 1000000 * 0xffff)
  File "/home/pi/.local/lib/python3.5/site-packages/adafruit_pca9685.py", line 71, in frequency
    return self._pca.frequency
  File "/home/pi/.local/lib/python3.5/site-packages/adafruit_pca9685.py", line 145, in frequency
    return self.reference_clock_speed / 4096 / self.prescale_reg
ZeroDivisionError: float division by zero
>>> `

It might happen cause of PCA gets clock speed as 0. 
What shall i do to fix this ? 
Thanks.

ladyada, 2019-10-31:
what is the exact script you're running?

OgiBalboa, 2019-10-31:
I tried running every example scripts and got same error. 
for instance : https://github.com/adafruit/Adafruit_CircuitPython_PCA9685/blob/master/examples/pca9685_servo.py
I checked adafruit servo and pca9685 modules and found the code caused this error : 
Look at the frequency.setter .
```python
class PCA9685:

    # Registers:
    mode1_reg = UnaryStruct(0x00, '<B')
    prescale_reg = UnaryStruct(0xFE, '<B')
    pwm_regs = StructArray(0x06, '<HH', 16)

    def __init__(self, i2c_bus, *, address=0x40, reference_clock_speed=25000000):
        self.i2c_device = i2c_device.I2CDevice(i2c_bus, address)
        self.channels = PCAChannels(self)
        """Sequence of 16 `PWMChannel` objects. One for each channel."""
        self.reference_clock_speed = reference_clock_speed
        """The reference clock speed in Hz."""
        self.reset()

    def reset(self):
        """Reset the chip."""
        self.mode1_reg = 0x00 # Mode1

    @property
    def frequency(self):
        """The overall PWM frequency in Hertz."""
        #return self.reference_clock_speed / 4096 / self.prescale_reg
        return self.reference_clock_speed / 4096 / self.ogibalboa

    @frequency.setter
    def frequency(self, freq):
        prescale = int(self.reference_clock_speed / 4096.0 / freq + 0.5)
        if prescale < 3:
            raise ValueError("PCA9685 cannot output at the given frequency")
        old_mode = self.mode1_reg # Mode 1
        self.mode1_reg = (old_mode & 0x7F) | 0x10 # Mode 1, sleep
        self.prescale_reg = prescale # Prescale  -- faulty variable ( its value returns 0)
        self.ogibalboa = prescale  # tried to assign to my own variable 
        self.mode1_reg = old_mode # Mode 1
        time.sleep(0.005)
        self.mode1_reg = old_mode | 0xa1 # Mode 1, autoincrement on
```
I checked  self.prescale_reg ' s value , it returned 0 but prescale returns 122 . When i assign prescale to another variable ( ogibalboa) its value is becomes 122.
After than code ran without error but servo doesnt move now.
Thanks.

caternuson, 2019-10-31:
Can you provide more details on your hardware setup. What are you using that has a PCA9685?

I am not able to recreate this issue running the example program from the PCA9685 library and the breakout.

[image]

```console
pi@pizerow:~ $ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: 40 -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         
pi@pizerow:~ $ python3 pca9685_servo.py 
pi@pizerow:~ $
```

OgiBalboa, 2019-11-01:
I am using PCA to control a servo. 1 servo connected to PCA and PCA connected to raspberry. 
I will check my hardware setup again when i will be able. But i did the same connection as in the picture. Tried to connect SDA and SCL false, to see reaction and program gave i2c error . So though I connected truely before.
But didnt do " i2cdetect -y 1 " Im going to try that. I turned on i2c from raspi config.
But if it is certainly not software caused problem, I will try to test my device on Arduino then ?

caternuson, 2019-11-01:
In general, this is sounding like some kind of hardware issue. Posting a photo of your setup would help. If the item with the PCA9685 is an Adafruit product, you can also post in the support forums:
https://forums.adafruit.com/

ladyada, 2019-11-01:
hiya agree that you should contact the vendor of your hardware for tech support as we can't replicate it here. please open if you find it is in fact a library issue!

closed 2019-11-01 by ladyada