adafruit/Adafruit_CircuitPython_ServoKit #20: PCA frequency not configurable

razvanphp, 2020-06-22:
Hello,

We're trying to control Traxxas digital servos and it seems the hardcoded 50 Hz frequency doesn't cut it. Can we maybe parametrise it?

Source: https://github.com/adafruit/Adafruit_CircuitPython_ServoKit/blob/master/adafruit_servokit.py#L92

Thank you!

caternuson, 2020-06-22:
Can you provide more information about the specific control signal that those servos use.

razvanphp, 2020-06-22:
the steering servo works fine with frequency set at 50, but the XL-5 ESC does not, I could make it work with frequency at above 100 but then at max I get a buffer error.

You can read some similar cases here: https://forums.traxxas.com/showthread.php?8923102-PWM-in-XL-5

razvanphp, 2020-06-24:
The error that we get when we increase the frequency:
```
File "webcam.py", line 42, in drive
steering.throttle = gamepad['axes'][0]
File "/home/jetson/.local/lib/python3.6/site-packages/adafruit_motor/servo.py", line 148, in throttle
self.fraction = (value + 1) / 2
File "/home/jetson/.local/lib/python3.6/site-packages/adafruit_motor/servo.py", line 72, in fraction
self._pwm_out.duty_cycle = duty_cycle
File "/home/jetson/.local/lib/python3.6/site-packages/adafruit_pca9685.py", line 98, in duty_cycle
self._pca.pwm_regs[self._index] = (0, value)
File "/home/jetson/.local/lib/python3.6/site-packages/adafruit_register/i2c_struct_array.py", line 75, in _setitem_
i2c.write(buf)
File "/home/jetson/.local/lib/python3.6/site-packages/adafruit_bus_device/i2c_device.py", line 104, in write
self.i2c.writeto(self.device_address, buf, start=start, end=end, stop=stop)
File "/home/jetson/.local/lib/python3.6/site-packages/busio.py", line 110, in writeto
return self._i2c.writeto(address, memoryview(buffer)[start:end], stop=stop)
File "/home/jetson/.local/lib/python3.6/site-packages/adafruit_blinka/microcontroller/generic_linux/i2c.py", line 49, in writeto
self._i2c_bus.write_bytes(address, buffer[start:end])
File "/home/jetson/.local/lib/python3.6/site-packages/Adafruit_PureIO/smbus.py", line 308, in write_bytes
self._device.write(buf)
OSError: [Errno 121] Remote I/O error
```

ladyada, 2020-06-24:
you likely have a power problem, not an i2c problem. use split supplies, etc.

dherrada, 2020-07-01:
@razvanphp Any luck figuring this out?

razvanphp, 2020-07-02:
not really, we are still trying to figure this out, it seems the adafruit board does not like to get the power from the PWM +5V pin itself? The Traxxas ESC does this to power the second steering servo. will try a second power source...

razvanphp, 2020-07-06:
ok, we figured it out, the `Remote I/O error` error was because of the wires, thank you for the tip.

But we still had to increase the frequency for the traxxas digital servos to work, so can we make this a parameter in the constructor please? I can push a PR also...

dherrada, 2020-07-10:
@razvanphp If you could make a PR, that'd be great.

caternuson, 2020-07-10:
Does it need to be a parameter in the constructor? Would it be OK to have it as a property?
```python
kit = ServoKit(channels=8)
kit.frequency = 100
```

tannewt, 2020-07-13:
You can always do `kit._pca.frequency = 50` if you want to use ServoKit. Otherwise, you can always use the underlying libraries directly.

I'm a little worried about adding features because it adds a bit of size and API surface to the library which is meant to be simple.

razvanphp, 2020-07-14:
> I'm a little worried about adding features because it adds a bit of size and API surface to the library which is meant to be simple.

I've pushed the PR, this is hardly increasing the complexity, let's be honest, it's just 3 lines of code...

`_pca` object should be considered private in this context.

tannewt, 2020-07-14:
Thanks for the PR! I agree it's not too bad.

I just want to make sure it's ok. Some folks want complexity in the *Kit libraries when they should be simple and the underlying libraries are the better place for complexity.

PR #21 (2020-07-14): Allow specifying the PCA frequency
Fixes #20
closed 2020-07-14 by tannewt