adafruit/Adafruit_CircuitPython_ServoKit #22: no module named 'adafruit_servokit'

DTX35, 2020-10-01:
Hi there, 

I have already used HAT or Bonnet channel PWM over the past few month and now I am configuring an new raspberry PI. 

I did follow the instructions here :https://learn.adafruit.com/adafruit-16-channel-pwm-servo-hat-for-raspberry-pi/using-the-python-library (I already enabled i2c, the raspberry can detect my board) 

And then I ran :sudo pip3 install adafruit-circuitpython-servokit
which gives me logs : 
`Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: adafruit-circuitpython-servokit in /usr/local/lib/python3.5/dist-packages (1.3.0)
Requirement already satisfied: adafruit-circuitpython-busdevice in /usr/local/lib/python3.5/dist-packages (from adafruit-circuitpython-servokit) (5.0.1)
Requirement already satisfied: Adafruit-Blinka in /usr/local/lib/python3.5/dist-packages (from adafruit-circuitpython-servokit) (5.4.0)
Requirement already satisfied: adafruit-circuitpython-motor in /usr/local/lib/python3.5/dist-packages (from adafruit-circuitpython-servokit) (3.2.3)
Requirement already satisfied: adafruit-circuitpython-pca9685 in /usr/local/lib/python3.5/dist-packages (from adafruit-circuitpython-servokit) (3.3.2)
Requirement already satisfied: adafruit-circuitpython-register in /usr/local/lib/python3.5/dist-packages (from adafruit-circuitpython-servokit) (1.9.0)
Requirement already satisfied: RPi.GPIO in /usr/lib/python3/dist-packages (from Adafruit-Blinka->adafruit-circuitpython-servokit) (0.6.5)
Requirement already satisfied: sysv-ipc in /usr/local/lib/python3.5/dist-packages (from Adafruit-Blinka->adafruit-circuitpython-servokit) (1.0.1)
Requirement already satisfied: rpi-ws281x>=4.0.0 in /usr/local/lib/python3.5/dist-packages (from Adafruit-Blinka->adafruit-circuitpython-servokit) (4.2.4)
Requirement already satisfied: Adafruit-PlatformDetect>=2.15.1 in /usr/local/lib/python3.5/dist-packages (from Adafruit-Blinka->adafruit-circuitpython-servokit) (2.17.0)
Requirement already satisfied: Adafruit-PureIO>=1.1.5 in /usr/local/lib/python3.5/dist-packages (from Adafruit-Blinka->adafruit-circuitpython-servokit) (1.1.5)
Requirement already satisfied: pyftdi>=0.40.0 in /usr/local/lib/python3.5/dist-packages (from Adafruit-Blinka->adafruit-circuitpython-servokit) (0.52.0)
Requirement already satisfied: pyserial>=3.0 in /usr/lib/python3/dist-packages (from pyftdi>=0.40.0->Adafruit-Blinka->adafruit-circuitpython-servokit) (3.2.1)
Requirement already satisfied: aenum>=2.1.0; python_version == "3.5" in /usr/local/lib/python3.5/dist-packages (from pyftdi>=0.40.0->Adafruit-Blinka->adafruit-circuitpython-servokit) (2.2.4)
Requirement already satisfied: pyusb>=1.0.0 in /usr/local/lib/python3.5/dist-packages (from pyftdi>=0.40.0->Adafruit-Blinka->adafruit-circuitpython-servokit) (1.1.0)
You are using pip version 19.0.3, however version 20.2.3 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
`

Which clearly make me think that the servokit module is installed but once I launch my little project, I get the following issue : 
Traceback (most recent call last):
  File "server.py", line 7, in <module>
    from servos import Servos
  File "/home/pi/backup excell/servos.py", line 9, in <module>
    from adafruit_servokit import ServoKit
ImportError: No module named 'adafruit_servokit'

Can someone help me please ? I cannot see what I did wrong

ladyada, 2020-10-01:
run  `pip3 install adafruit-circuitpython-servokit` too

DTX35, 2020-10-01:
Hello ladyada, 

THnak you for your reply, 
As mentioned I already run sudo pip3 install adafruit-circuitpython-servokit

but just to be sure did just run pip3 install adafruit-circuitpython-servokit as you said but stil got the same issue.

ladyada, 2020-10-01:
something is weird about your python setup...but not sure what. sorry!

caternuson, 2020-10-01:
Also make sure you are running your script with Python3, not 2?

DTX35, 2020-10-01:
It seems that my python version is 3.5 which happens to be deprecated since september. I am not sure that this is why I have my modle error but I will try to update python first

AK51, 2020-10-09:
I am using Nvidia Xavier NX board with PCA9685.
I just used the example code, but the line kit=ServoKit(channels=16), it gives error said ModuleNotFoundError: No module named 'machine'.
The error starts from adafriut_servokit.py --> ..python3.6/site-packages/busio.py...

caternuson, 2020-10-09:
@AK51 Please start a new issue. This one relates to Pi.

Govind-pixel, 2020-10-12:
Hi AK51,
I am using Nvidia Jetson nano with PCA9685 I am also getting same error like you.Is there any fix available for this.

closed 2020-10-15 by ladyada

SYTECH27, 2022-07-30:
I have a problem with importing the name Servokit. It states: Python 3.6.9 (default, Mar 15 2022, 13:55:28) 
[GCC 8.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from adafruit_servokit import Servokit
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: cannot import name 'Servokit'
Can you please help me with this problem if you can that would be great.

caternuson, 2022-07-30:
need capital K in ServoKit
https://github.com/adafruit/Adafruit_CircuitPython_ServoKit/blob/54721199a137dab7224237bf168aedb62b83707f/examples/servokit_simpletest.py#L6

SYTECH27, 2022-07-31:
Thank you for your response. It was a silly mistake, but thank you for the
help!