adafruit/Adafruit_CircuitPython_ServoKit #23: Problem with ServoKit in Nvidia Xavier NX board.

AK51, 2020-10-09:
I am using Nvidia Xavier NX board with PCA9685. Not sure if ServoKit has updated the version? Or The Nvidia Xavier NX Jetpack 4.4 does not support ServoKit library anymore...
I followed this youtube and his code works.
https://youtu.be/JSlJ8NVVWfQ?list=PLGs0VKk2DiYw_Xx7c3M_9HMEcn9hPhFJV&t=2773
I just used the example code, but in the line kit=ServoKit(channels=16), it gives error said ModuleNotFoundError: No module named 'machine'.
The error starts from adafriut_servokit.py --> ..python3.6/site-packages/busio.py...
Thx.
[image]

ErickDiaz, 2020-10-11:
I'm getting the same issue, I have recently updated the libraries, I didn't have the issue before the update, I started a fresh new install of python in the jetson nano and had the issue again
```
Initializing Servos
Traceback (most recent call last):
  File "spotmicro_ai/jetson_nano/src/spotmicroai/initial_position.py", line 11, in <module>
    i2c_bus0=(busio.I2C(board.SCL_1, board.SDA_1))
  File "/home/balam/balam-env/lib/python3.6/site-packages/busio.py", line 31, in __init__
    self.init(scl, sda, frequency)
  File "/home/balam/balam-env/lib/python3.6/site-packages/busio.py", line 59, in init
    from machine import I2C as _I2C
ModuleNotFoundError: No module named 'machine'
Exiting... 
```

Govind-pixel, 2020-10-12:
Hi All,
I am using Nvidia Jetson nano with PCA9685 I am also getting same error like you.Is there any fix available for this.

ladyada, 2020-10-12:
@swarren has a fix coming

closed 2020-10-15 by ladyada