adafruit/Adafruit_CircuitPython_ServoKit #24: Angle-Servo Moves but Continuous servo motor doesn't move

mhaboali, 2020-10-15:
**Describe the issue**
I have one servo motor is connected to the first channel and another continuous motor is connected to channel 2 to the pca_9685

I've set up this library on my Jetson Xavier Nx and I managed to command an angle-servo motor but when I've commanded a continuous servo motor, it didn't move.

**NVIDIA Jetson Hardware**
- [ X] Jetson Xavier Nx.

I"ve used this code
```
from adafruit_servokit import ServoKit
import board
import busio
import time
print("Initializing Servos")
Initializing Servos
i2c_bus0=(busio.I2C(board.SCL_1, board.SDA_1))
print("Initializing ServoKit")
Initializing ServoKit
kit = ServoKit(channels=16, address=0x41, i2c=i2c_bus0)
```
And when use this `kit.servo[0].angle=30` the motor moves but when use this command `kit.continuous_servo[1].throttle=0.5` the motor doesn't move.

Any help, please?

ladyada, 2020-10-15:
are you following a youtuber video? if so please ask them for support with their tutorial series

mhaboali, 2020-10-15:
No, I just followed the PythonCircuit tutorial on Adafruit tutorials.

caternuson, 2022-04-20:
This one is a bit old, but just tested and can not recreate. Closing for now.

With a [continuous rotation servo](https://www.adafruit.com/product/3614) attached to output channel 1 of a [PCA9685 breakout](https://www.adafruit.com/product/815) running the following code:
```python
Adafruit CircuitPython 7.2.5 on 2022-04-06; Adafruit QT Py M0 with samd21e18
>>> from adafruit_servokit import ServoKit
>>> kit = ServoKit(channels=16)
>>> kit.continuous_servo[1].throttle = 0
>>> kit.continuous_servo[1].throttle = 0.1
>>> kit.continuous_servo[1].throttle = 0.2
>>> kit.continuous_servo[1].throttle = 0.5
>>> kit.continuous_servo[1].throttle = 1
>>>  
```
servo behaves as expected.

closed 2022-04-20 by caternuson