adafruit/Adafruit_CircuitPython_ServoKit #25: sphinx does not properly link to adafruit_motor classes

ktdreyer, 2021-01-01:
Something is not quite right with the Sphinx links for adafruit_motor. The [servokit API page](https://circuitpython.readthedocs.io/projects/servokit/en/latest/api.html) has the strings `:py:class:~adafruit_motor.servo.ContinuousServo` and `:py:class:~adafruit_motor.servo.Servo` rather than linking to the class docs for `ContinuousServo` and `Servo`.

I was looking at this a bit, and I can't figure out how to make Sphinx link those classes, but maybe someone has some idea?

ladyada, 2021-01-02:
im not a sphinx-pert... @tannewt any advice (or someone else who knows the deets)?

tannewt, 2021-01-05:
I think there are two things to do:
1. Use one ` instead of two where the references are. For example: https://github.com/adafruit/Adafruit_CircuitPython_ServoKit/blob/master/adafruit_servokit.py#L107 Once that's done, the doc build should break.
2. Add the intersphinx mapping for motor to the conf.py here: https://github.com/adafruit/Adafruit_CircuitPython_ServoKit/blob/master/docs/conf.py#L33 It should follow the same pattern as Register. This should fix the doc build.

@ktdreyer Want to try fixing it?

jposada202020, 2021-04-23:
Fix by #28

closed 2021-04-23 by jposada202020