adafruit/Adafruit_CircuitPython_ServoKit #27: continuous never stops

rchadgray, 2021-03-25:
Hello,

so far my servo that is labeled continuous never stops.

This makes it turn CCW:  
kit.continuous_servo[1].throttle = 1

This turns it CW
kit.continuous_servo[1].throttle = -1

This slows the CW spinning
kit.continuous_servo[1].throttle = 0

the python exits and the servo keeps running.

ladyada, 2021-03-25:
you have to 'center' the servo - contact the servo seller for how to do that! :)

closed 2021-03-25 by ladyada

rchadgray, 2021-03-25:
Ah!!!  never knew that.  Makes sense.

Thanks!

rchadgray, 2021-03-26:
Found it for my servo.  https://www.pololu.com/product/2820

The FS90R continuous rotation servo converts standard RC servo position pulses into continuous rotation speed. The default rest point is 1.5 ms, but this can be adjusted by using a small screwdriver to turn the middle-point adjustment potentiometer. Pulse widths above the rest point result in counterclockwise rotation, with speed increasing as the pulse width increases; pulse widths below the rest point result in clockwise rotation, with speed increasing as the pulse width decreases.

Now i think the adafruit tutorial talks about converting ms to frequency... so i will go back and read that.

Thanks again!
Chad