adafruit/Adafruit_CircuitPython_ServoKit #29: Servos stuck when run servokit_all_servos_synchronised.

DarkBasement, 2021-05-01:
Hi, I run this example: servokit_all_servos_synchronised.py to rotate several servos MG996R through PCA9685 but it works partially. Sometimes, some of the servos stop moving and stop reacting to the angle changes. They stuck. Servos can do it by one or in couples. More servos I use - more probability of stopping. I use external power so I don't think that it's power-related.

I follow servokit_all_servos_synchronised example. If I connect 6 servos then this code will make at least one servo unresponsive during the first run. During the second run there may be one more unresponsive servo appear.
for i in range(len(kit.servo)):
    kit.servo[i].angle = 180

I noticed that if I add delay then it works fine. For example:
for i in range(len(kit.servo)):
    kit.servo[i].angle = 180
    time.sleep(0.2)

0.2s is a small value that makes the servos move together but not exactly how I want it.
Do you have any idea why this happens? Is this example valid for PCA9685 ?

ladyada, 2021-05-01:
what is the exact hardware you are using? please post the exact urls or photos

DarkBasement, 2021-05-13:
Thank you for pointing to the hardware. I double-checked my mg966r servos and some of them appeared to be defective. They worked alone but refused to work in a group like I described before. I replaced them and it works now. I am able to control 12 mg966r servos simultaneously.

closed 2021-05-13 by DarkBasement