adafruit/Adafruit_CircuitPython_ServoKit #30: can't release servo's after using them

high-solutions, 2021-05-14:
What ever I tried, i can't find a way to release the servo's once they have been used without pulling the power of the PCA9585 board. But after pulling power and putting it back the servo's keep release until they are activated again.

ladyada, 2021-05-15:
Try setting the angle to `None`

n8atnite, 2022-02-26:
This does not work for me. Servos remain locked even after setting all channels' angle to `None`.

high-solutions, 2022-02-27:
Yep, i solved the issue by connecting the extra pin but that enables/disables all servo's instead of only one. Haven't found the reall solution yet

caternuson, 2022-04-18:
Using the `OE` pin is always an option. That'd be the hardware way to disable *all* the outputs. But should be able to disable per channel via software as well. Just need to set duty cycle to `0` to essentially flat line the PWM signal, which should cause the servo to be released. And as suggested above, that should be possible by setting the angle to `None`.

Just tested this with a [PCA breakout](https://www.adafruit.com/product/815) (the PCA9685 is what all the servo HATs, Bonnets, etc. use):
```python
Adafruit CircuitPython 7.2.5 on 2022-04-06; Adafruit QT Py M0 with samd21e18
>>> from adafruit_servokit import ServoKit
>>> kit = ServoKit(channels=8)
>>> kit.servo[0].angle = 0
>>> kit.servo[0].angle = 90
>>> kit.servo[0].angle = 180
>>> kit.servo[0].angle = None
>>>  
```
and it did release the servo after setting `None`. There was a small delay before the servo actually released, but it eventually (1 second-ish) became free and was turnable by hand.

If you try the above same code snippet, the servo still does not release on your setup?

dattasaurabh82, 2022-04-20:
@caternuson `None` works

caternuson, 2022-04-20:
OK. gonna close this then. Original poster found another solution. And otherwise can not recreate the issue. Setting angle to `None` is how this should be done.

Additional background info for reference. The ability to release a servo by setting angle to `None` was added in this PR:
https://github.com/adafruit/Adafruit_CircuitPython_Motor/pull/27
This ServoKit library is just a thin wrapper layer over the Motor library to make the Servo use case easier for some specific products. But it generally calls directly into the Motor library.

closed 2022-04-20 by caternuson