adafruit/Adafruit_CircuitPython_ServoKit #38: Added type annotations on latest update is not backwards compatible with python3.6>

NateJose123, 2022-04-27:
Got this error:

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/usr/lib/python3.5/py_compile.py", line 129, in compile
    raise py_exc
py_compile.PyCompileError:   File "./prog.py", line 1
    from __future__ import annotations
                                     ^
SyntaxError: future feature annotations is not defined

Fixed it by upgrading python but would be great to have an update that is backwards compatible with python 3.6 and below if possible. Thanks!

dhalbert, 2022-04-27:
Unfortunately, `from __future__ import ...` has to be the very first statement in a Python script, and so it cannot be guarded with `try`-`except` or anything like that. So we have upped the requirements to Python 3.7 for most things. Python 3.6 is end-of-life, as of four months ago.

tekktrik, 2022-04-27:
@dhalbert Is this specific to this library?  Happy to add Blinka>=7.0.0 as a dependency if so, and add a major up-rev if that would help.

ladyada, 2022-06-05:
@tekktrik yah go ahead and update the requirements/deps