adafruit/Adafruit_CircuitPython_ServoKit #40: Can't release servo using Adafruit PCA9685 16-Channel Servo Driver

ananasn, 2022-05-31:
Hello!

I recently got the same issue as the topic starter. Still can't get around with it.

I got the `Adafruit PCA9685 16-Channel Servo Driver` connected to `Raspberry Pi 4`.

On my RPI following packages are installed in venv:
```
Adafruit-Blinka==7.3.0
adafruit-circuitpython-busdevice==5.1.8
adafruit-circuitpython-motor==3.3.5
adafruit-circuitpython-pca9685==3.4.1
adafruit-circuitpython-register==1.9.8
adafruit-circuitpython-servokit==1.3.8
adafruit-circuitpython-typing==1.7.0
Adafruit-PlatformDetect==3.22.1
Adafruit-PureIO==1.1.9
install==1.3.5
pkg-resources==0.0.0
pyftdi==0.54.0
pyserial==3.5
pyusb==1.2.1
rpi-ws281x==4.3.4
RPi.GPIO==0.7.1
sysv-ipc==1.1.0
typing-extensions==4.2.0
```

My python code is following:
```python3
from adafruit_servokit import ServoKit
import time

kit = ServoKit(channels=16)
kit.servo[0].angle = None
time.sleep(10)
```

During code execution and after it finish my servo keeps blocked.

_Originally posted by @ananasn in https://github.com/adafruit/Adafruit_CircuitPython_ServoKit/issues/30#issuecomment-1142516685_

caternuson, 2022-06-06:
Still can't recreate this. It works on test servo here.

@ananasn Can you link to the specific servo you are using.

ananasn, 2022-06-19:
I use [these](https://aliexpress.ru/item/32907625266.html?gatewayAdapt=glo2rus&item_id=32907625266&sku_id=65846679148&spm=a2g0s.12269583.0.0.2147789eSjIPVu) servos in my project.

ladyada, 2022-06-20:
can you move the servo BEFORE running the script?

caternuson, 2022-09-28:
Closing due to lack of response. Could not recreate issue. Potentially specific to servos being used.

closed 2022-09-28 by caternuson

scprotz, 2022-12-07:
I am having the same problem with different brands of servos and the PCA9685 using CircuitPython ServoKit.  Some brands of servos 'release' on servo.angle = None and other brands do not.  Why would a servo state locked/on even when angle is set to None?  Is the PCA9685 still sending signal to the server at that point?  Is there an alternate way to cut the signal to release the servo completely?