adafruit/Adafruit_CircuitPython_ServoKit #42: Could not find a version that satisfies the requirement Adafruit-Blinka>=7.0.0

Real4LA, 2022-06-19:
hey im trying to install adafruit-circuitpython-servokit but i keep getting this error:
Collecting Adafruit-Blinka>=7.0.0 (from adafruit-circuitpython-servokit)
  Could not find a version that satisfies the requirement Adafruit-Blinka>=7.0.0 (from adafruit-circuitpython-servokit) (from versions: 0.1.3, 0.1.4, 0.1.5, 0.1.6, 0.1.7, 0.1.8, 0.1.9, 0.1.10, 0.2, 0.2.1, 0.2.2, 0.2.3, 0.2.4, 0.2.5, 0.2.6, 0.2.7, 0.3.0, 0.3.1, 0.3.2, 0.4.0, 1.0.2, 1.1.0, 1.1.1, 1.2.0, 1.2.1, 1.2.3, 1.2.4, 1.2.5, 1.2.6, 1.2.7, 1.2.8, 1.3.0, 1.3.1, 1.3.2, 1.3.3, 1.3.4, 2.0.0, 2.0.1, 2.0.2, 2.1.0, 2.1.1, 2.1.2, 2.1.3, 2.1.4, 2.1.5, 2.1.6, 2.2.0, 2.2.1, 2.3.0, 2.3.1, 2.3.2, 2.3.3, 2.4.0, 2.4.1, 2.5.0, 2.5.1, 2.5.2, 2.5.3, 2.6.0, 2.6.1, 3.0.0, 3.0.1, 3.0.2, 3.0.3, 3.0.4, 3.0.5, 3.0.6, 3.0.7, 3.0.8, 3.1.0, 3.1.1, 3.1.2, 3.2.0, 3.3.0, 3.3.1, 3.3.2, 3.3.3, 3.3.4, 3.3.5, 3.3.6, 3.3.7, 3.3.8, 3.3.9, 3.3.10, 3.4.0, 3.4.1, 3.5.0, 3.5.1, 3.5.2, 3.6.0, 3.6.1, 3.7.0, 3.7.1, 3.8.0, 3.9.0, 3.10.0, 4.0.0, 4.1.0, 4.2.0, 4.3.0, 4.4.0, 4.5.0, 4.6.0, 4.7.0, 4.8.0, 4.8.1, 4.9.0, 4.10.0, 4.10.1, 5.0.0, 5.0.1, 5.1.0, 5.2.0, 5.2.1, 5.2.2, 5.2.3, 5.2.4, 5.3.0, 5.3.1, 5.3.2, 5.3.3, 5.3.4, 5.4.0, 5.4.1, 5.5.0, 5.5.1, 5.5.2, 5.5.3, 5.5.4, 5.6.0, 5.7.0, 5.8.0, 5.8.1, 5.8.2, 5.9.0, 5.9.1, 5.9.2, 5.10.0, 5.11.0, 5.12.0, 5.13.0, 5.13.1, 6.0.0, 6.0.1, 6.0.2, 6.1.0, 6.2.0, 6.2.1, 6.2.2, 6.3.0, 6.3.1, 6.3.2, 6.4.0, 6.4.1, 6.4.2, 6.5.0, 6.6.0, 6.6.1, 6.6.2, 6.7.0, 6.8.0, 6.8.1, 6.8.2, 6.8.3, 6.9.0, 6.9.1, 6.9.2, 6.10.0, 6.10.1, 6.10.2, 6.10.3, 6.11.0, 6.11.1, 6.12.0, 6.13.0, 6.13.1, 6.14.0, 6.14.1, 6.15.0)
No matching distribution found for Adafruit-Blinka>=7.0.0 (from adafruit-circuitpython-servokit)

Neradoc, 2022-06-19:
Blinka 7 requires python 3.7 or later, are you running an earlier version of python ?

Real4LA, 2022-06-19:
> 

> Blinka 7 requires python 3.7 or later, are you running an earlier version of python ?

yes 3.6

ladyada, 2022-06-20:
please check the jetson support forums for help, we know others have gotten it working - we do not own a jetson

closed 2022-06-20 by ladyada