adafruit/Adafruit_CircuitPython_ServoKit #44: No module named '_sysconfigdata_m_linux_aarch64-linux-gnu'

Real4LA, 2022-06-20:
jetson@jetson:~$ pip3 install adafruit-circuitpython-servokit

Collecting adafruit-circuitpython-servokit
  Using cached https://files.pythonhosted.org/packages/a2/65/5da8d1d4a22701fedf9dbae9094e40185d58be25b86175eb0861d55398a5/adafruit-circuitpython-servokit-1.3.10.tar.gz
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-build-oy6aixzt/adafruit-circuitpython-servokit/setup.py", line 61, in <module>
        py_modules=["adafruit_servokit"],
      File "/usr/lib/python3/dist-packages/setuptools/__init__.py", line 128, in setup
        _install_setup_requires(attrs)
      File "/usr/lib/python3/dist-packages/setuptools/__init__.py", line 123, in _install_setup_requires
        dist.fetch_build_eggs(dist.setup_requires)
      File "/usr/lib/python3/dist-packages/setuptools/dist.py", line 513, in fetch_build_eggs
        replace_conflicting=True,
      File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 774, in resolve
        replace_conflicting=replace_conflicting
      File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 1057, in best_match
        return self.obtain(req, installer)
      File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 1069, in obtain
        return installer(requirement)
      File "/usr/lib/python3/dist-packages/setuptools/dist.py", line 579, in fetch_build_egg
        cmd.ensure_finalized()
      File "/usr/lib/python3.7/distutils/cmd.py", line 107, in ensure_finalized
        self.finalize_options()
      File "/usr/lib/python3/dist-packages/setuptools/command/easy_install.py", line 292, in finalize_options
        'site_dirs',
      File "/usr/lib/python3/dist-packages/setuptools/command/easy_install.py", line 1402, in _expand
        config_vars = self.get_finalized_command('install').config_vars
      File "/usr/lib/python3.7/distutils/cmd.py", line 299, in get_finalized_command
        cmd_obj.ensure_finalized()
      File "/usr/lib/python3.7/distutils/cmd.py", line 107, in ensure_finalized
        self.finalize_options()
      File "/usr/lib/python3/dist-packages/setuptools/command/install.py", line 38, in finalize_options
        orig.install.finalize_options(self)
      File "/usr/lib/python3.7/distutils/command/install.py", line 306, in finalize_options
        (prefix, exec_prefix) = get_config_vars('prefix', 'exec_prefix')
      File "/usr/lib/python3.7/distutils/sysconfig.py", line 508, in get_config_vars
        func()
      File "/usr/lib/python3.7/distutils/sysconfig.py", line 468, in _init_posix
        _temp = __import__(name, globals(), locals(), ['build_time_vars'], 0)
    ModuleNotFoundError: No module named '_sysconfigdata_m_linux_aarch64-linux-gnu'

    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-oy6aixzt/adafruit-circuitpython-servokit/

dhalbert, 2022-06-20:
This is on a Jetson? My understanding is that the the default Python on a Jetson is earlier than 3.7. It looks like you might have installed Python 3.7 by hand, but that there are some things missing.

Real4LA, 2022-06-20:
> This is on a Jetson? My understanding is that the the default Python on a Jetson is earlier than 3.7. It looks like you might have installed Python 3.7 by hand, but that there are some things missing.

it is on jetson i turned py3.7 to default and now am getting this do u have any idea how to fix it ?
Exception:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/pip/basecommand.py", line 215, in main
    status = self.run(options, args)
  File "/usr/lib/python3/dist-packages/pip/commands/install.py", line 368, in run
    isolated=options.isolated_mode,
  File "/usr/lib/python3/dist-packages/pip/commands/install.py", line 454, in get_lib_location_guesses
    scheme = distutils_scheme('', *args, **kwargs)
  File "/usr/lib/python3/dist-packages/pip/locations.py", line 153, in distutils_scheme
    i.finalize_options()
  File "/usr/share/python-wheels/setuptools-39.0.1-py2.py3-none-any.whl/setuptools/command/install.py", line 38, in finalize_options
    orig.install.finalize_options(self)
  File "/usr/lib/python3.7/distutils/command/install.py", line 390, in finalize_options
    ('build_lib', 'build_lib'))
  File "/usr/lib/python3.7/distutils/cmd.py", line 286, in set_undefined_options
    src_cmd_obj = self.distribution.get_command_obj(src_cmd)
  File "/usr/lib/python3.7/distutils/dist.py", line 857, in get_command_obj
    klass = self.get_command_class(command)
  File "/usr/share/python-wheels/setuptools-39.0.1-py2.py3-none-any.whl/setuptools/dist.py", line 634, in get_command_class
    self.cmdclass[command] = cmdclass = ep.load()
  File "/usr/share/python-wheels/pkg_resources-0.0.0-py2.py3-none-any.whl/pkg_resources/__init__.py", line 2324, in load
    return self.resolve()
  File "/usr/share/python-wheels/pkg_resources-0.0.0-py2.py3-none-any.whl/pkg_resources/__init__.py", line 2330, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
ModuleNotFoundError: No module named 'setuptools.command.build'

dhalbert, 2022-06-20:
It looks like you only have part of a python3.7 installation: you're missing setuptools and some other things. I would not necessarily recommend replacing the system python with 3.7. Instead,  you can install a private copy somewhere else and have it be more complete, and then use `venv` or similar to create a 3.7 environment to run in. I have seen descriptions of installing 3.7 on Jetson in the Jetson forums and some other places.

ladyada, 2022-06-20:
https://stackoverflow.com/questions/14426491/python-3-importerror-no-module-named-setuptools

caternuson, 2024-06-21:
Closing. This is not a library issue. It is a Python system setup issue.

closed 2024-06-21 by caternuson